bastibl/instant-gnuradio #1: VM, per the instructions, missing from the download location

PeterCielos, 2018-03-14:
Bastille,

This is great work thank you for posting, but per the instructions, the download link for the VM doesn't work as you can see from here:

> If you just want to try the VM, you can download it from here and import it into your VM appliance.[https://www.fleark.de/instant-gnuradio.ova](url)

Can the VM "instant-gnuradio.ova" be downloaded from another location ?

bastibl, 2018-03-17:
Fixed by deleting the download link from the readme :-)

No just kidding, but I did not yet find a good solution to automate the build and upload the image. Currently, it takes me hours. So the most recent version will not always be available to download.

Anyhow, the image I used at WLPC is available at http://fleark.de/instant-gnuradio-wlpc.ova

PeterCielos, 2018-03-17:
Bastille,

> Fixed by deleting the download link from the readme :-) 

Thank you I downloaded so this issue is resolved.

closed 2018-03-17 by PeterCielos