bastibl/instant-gnuradio #3: Virtualbox does not recognize --audioout and --audioin options

jkbecker, 2018-03-23:
I could only get the packer to work after modifying `base.json` as follows:

``` 
diff --git a/base.json b/base.json
index 8c8f861..1429781 100644
--- a/base.json
+++ b/base.json
@@ -39,9 +39,7 @@
             ["modifyvm", "{{.Name}}", "--usb", "on"],
             ["modifyvm", "{{.Name}}", "--usbxhci", "on"],
             ["modifyvm", "{{.Name}}", "--clipboard", "bidirectional"],
-            ["modifyvm", "{{.Name}}", "--audio", "pulse"],
-            ["modifyvm", "{{.Name}}", "--audioout", "on"],
-            ["modifyvm", "{{.Name}}", "--audioin", "on"]
+            ["modifyvm", "{{.Name}}", "--audio", "pulse"]
         ]
     }],
     "provisioners": [
``` 

Haven't been able to test whether this impacts the functionality at all; I'm getting a https certificate error on the virtualbox homepage that prevents me from looking up any details...

I thought I'd just leave this here and update as I test (hope that's ok).

jkbecker, 2018-03-24:
This is a problem, as the VM will not have audio support as it seems (who would have thought...)

bastibl, 2018-03-24:
I guess you have an older version of VirtualBox. I just installed the most recent one from the website (for Ubuntu 17.10). There VBoxManage has the option.
Anyhow, it only enables audio output by default, saving you one click. The device and everything is there nevertheless. It is just muted by default if you delete these options.

jkbecker, 2018-03-24:
Interesting. I was on virtualbox 5.0.4, moving to 5.2.8 solved it indeed. Thanks for the pointer!

closed 2018-03-24 by jkbecker