bastibl/instant-gnuradio #4: Default VM does not recognize USRP B200 plugged into the host machine

jkbecker, 2018-03-23:
Title says it all. The device is properly recognized on the host side but does not show up on the VM.

Is there anything that I have to configure first to make this work? If so, I'll gladly contribute to the documentation as I set this up; it looks like I'll be using this for at least a few immediate experiments.

bastibl, 2018-03-23:
Two things: if you use Virtualbox, you'll have to install the extension pack. Then you'll have to forward the SDR in the VM, lust like any other USB device. In the context menu, it should be something like Devices -> USB -> [B210].

jkbecker, 2018-03-23:
Ugh, sorry, this was premature; restarted everything (including my host machine) and it worked. I guess the virtualbox user group was not properly registered the first time.
Thanks for the quick reply; closing.

closed 2018-03-23 by jkbecker