bastibl/instant-gnuradio #8: Could you share a prebuilt USB image?

daniloncmayo, 2018-03-28:
Title says everything, could you share a pre built usb image?
That will help people that does not only to customize but rather just have a live environment.

I would be happy to seed 24/7 a torrent image...

Thanks!

daniloncmayo, 2018-03-28:
Replying myself, I just checked that the issue #1 has a link to the image.
Thanks

closed 2018-03-28 by daniloncmayo

catalinalb, 2019-09-19:
Dose anyone have a prebuild usb image if so please share it. I still have trouble creating one. Thank you.

bastibl, 2019-09-19:
Here is the link to an ISO: https://www.nocolour.de/files/instant-gnuradio.iso

catalinalb, 2019-09-19:
Thank you for the ISO.
Can you share the username and password as well? I did tryed gnuradio for user and password but didn't worked.

bastibl, 2019-09-19:
actually that should be the correct credentials. I didn't build the ISO, but I don't expect that something was changed. The only thing that comes to my mind is an empty password.

catalinalb, 2019-09-20:
Hi, tryed but no pass. I also tryed the usual ones like admin/admin, root/toor, gnuradio with and without password no pass. 😐

bastibl, 2019-09-20:
I don't know how this image was generated. You could just use the VM (I'd recommend that) or build your own iso. It's not really a reliable way to generated one, but basic instructions are here:
https://www.bastibl.net/instant-gnuradio/

catalinalb, 2019-09-22:
Dear Bastian,
I respect and appreciate your work and your achievements into RF world. I did spend this weekend trying to get it running still I run into potholes. I have a minimal ubuntu 18.04 fresh install dedicated specially for this (username and password gnuradio). I have added to the setup-gnuradio.sh script the missing line for GMP (not sure if it was really line 45...).
The make command finishes with "Build finished but no artifacts were created" Status 225 and error 1
I did run it again the same messages appear and up a few rows up is: "error in process sentinel: Window 3 too small for splitting ..... Unregistering and deleting virtual maschine....

don´t know how and what to do next so I prepared the USB stick I have to change your commands to sdb since I can´t/don´t know how to change the internal hard drive to sdb so that the stick can be sda. I have read the topic with the udev.rules I did modify it restart and nothing changed.

the keyboard is set to US and I always started the terminal with sudo bash.... so I did run all the commands as superuser.

Question: If I download the ova that you provide where do I have to move it so that I can use the generate iso command using your ova file? is it possible?

Thank you for your patience. I am just a novice in linux ... still learning by doing. 
If you have any ideea please let me know. 

Thank you,
Catalin