bastibl/instant-gnuradio #10: Trouble Building the VM

cyrus104, 2018-06-23:
Thank you for doing all the leg work for us on this one. I'm having a hard time building it at the moment.

I clone your repo then run ./configure and make. I see the VM being created that apt packages being install but then I get the error below.

[image]

bastibl, 2018-06-24:
I'm having issues with this from time to time :-/ I just pushed some changes so that a fixed version of Spacemacs is checked out (instead of the development version). Could you give it another try?

cyrus104, 2018-06-24:
So far so good, I'm on to building pybombs! Will let you know when it's done.

cyrus104, 2018-06-24:
Well, I made it much farther but now it looks like an opencl issue for fospher.

[image]

bastibl, 2018-06-24:
grrr, sorry for this. I rebuilt the whole thing today without problems :-/

Since there is no Fosphor `build` directory, the error suggests that there were previous issues installing Fosphor through PyBombs. (It seems like PyBombs doesn't return a non-zero exit status if something failed. So the installation process might have continued and failed at a later stage.)

cyrus104, 2018-06-24:
I'm in the process of rebuilding it now. Hopefully it will get a little farther.

I didn't think of it before but I need to make a change to prevent a deleting of the vm if it fails.

cyrus104, 2018-06-26:
I'm still working through it but looks like there might be an error that I'm having is with the mirror.

[image]

cyrus104, 2018-06-26:
I know I took the hard way but I added a file asset that was the US sources.list and then added a "sudo cp -f assets/sources.list /etc/apt/sources.list" to the init-setup.sh file.

This allowed me to successfully build the VM, I'm not sure why the default install is using the DE archive ubuntu servers (since they are all rysnc mirrored it shouldn't matter).
Thank you for taking a look at this for me.

bastibl, 2018-06-27:
Thanks, good to know. I had the same problems also with other mirrors and, IIRC, the generic mirror was incredible slow. So I'm not really sure how to fix that :-/

cyrus104, 2018-06-28:
Yeah, for home and office I keep a mirror of the repo and keep it pointed to the generic so my local speed is super fast.

I know it's not your build but with an ettus and hackrf, I can do a test listening to FM on gqrx but when I do a basic test in GnuRadio I get a bunch of audio underruns (aU).

cyrus104, 2018-08-24:
This build has been super useful but I wanted to see about changing the default repo to either the default or one can be easily be changed. For some reason, whenever I do an update and forget to merge my code the "default" repo in this build always have file size issues. As soon as a I change the repo in the vm and then build out gnuradio it works fine.

Thanks!

cyrus104, 2018-08-24:
I had to remove L8 proxy line too but after that it's working. Pulling from a repo on the LAN is so much faster for building.

Thank you for pointing out those lines.

closed 2018-08-24 by cyrus104