bastibl/instant-gnuradio #11: Invalid --audio argument 'pulse'

deusama, 2018-07-04:
I just wanted to test the script and I run into the following error on OSX:

`
Build 'virtualbox-iso' errored: Error executing command: VBoxManage error: VBoxManage: error: Invalid --audio argument 'pulse'
`

Latest version of Virtualbox 5.2.12 r122591 installed on my machine. Here comes the full output:

`
alfa:instant-gnuradio alfa$ ./configure

Checking for wget (tested with 1.17.1)
GNU Wget 1.19.5 built on darwin17.5.0.

-cares +digest -gpgme +https +ipv6 +iri +large-file -metalink +nls
+ntlm +opie +psl +ssl/gnutls

Wgetrc:
    /opt/local/etc/wgetrc (system)
Locale:
    /opt/local/share/locale
Compile:
    /usr/bin/clang -DHAVE_CONFIG_H
    -DSYSTEM_WGETRC="/opt/local/etc/wgetrc"
    -DLOCALEDIR="/opt/local/share/locale" -I. -I../lib -I../lib
    -I/opt/local/include -DNDEBUG -pipe -Os -arch x86_64
Link:
    /usr/bin/clang -DNDEBUG -pipe -Os -arch x86_64 -L/opt/local/lib
    -Wl,-headerpad_max_install_names -arch x86_64 -L/opt/local/lib
    -lpcre -lidn2 -lnettle /opt/local/lib/libgnutls.dylib
    -L/opt/local/lib -lz -L/opt/local/lib -lpsl ftp-opie.o gnutls.o
    http-ntlm.o ../lib/libgnu.a /opt/local/lib/libiconv.dylib
    /opt/local/lib/libintl.dylib -Wl,-framework -Wl,CoreFoundation
    /opt/local/lib/libunistring.dylib

Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later
<http://www.gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Originally written by Hrvoje Niksic <hniksic@xemacs.org>.
Please send bug reports and questions to <bug-wget@gnu.org>.

Checking for HashiCorp Packer (tested with 1.2.1)
1.2.4

Checking for Oracle VM VirtualBox (tested with 5.2.8)
5.2.12r122591

Checking for Oracle VM VirtualBox Extension Pack (tested with 5.2.8)
Pack no. 0:   Oracle VM VirtualBox Extension Pack

Great! Looks like all dependencies are installed. You can go ahead and type 'make' to build the VM.
alfa:instant-gnuradio alfa$ make
cd assets && wget http://registrationcenter-download.intel.com/akdlm/irc_nas/12556/opencl_runtime_16.1.2_x64_rh_6.4.0.37.tgz
--2018-07-04 13:37:48--  http://registrationcenter-download.intel.com/akdlm/irc_nas/12556/opencl_runtime_16.1.2_x64_rh_6.4.0.37.tgz
Resolving registrationcenter-download.intel.com (registrationcenter-download.intel.com)... 88.221.235.4, 88.221.235.58
Connecting to registrationcenter-download.intel.com (registrationcenter-download.intel.com)|88.221.235.4|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 117485559 (112M) [application/x-compressed]
Saving to: 'opencl_runtime_16.1.2_x64_rh_6.4.0.37.tgz'

opencl_runtime_16.1.2_x64_rh_6.4.0.37. 100%[===========================================================================>] 112.04M  5.27MB/s    in 31s

2018-07-04 13:38:20 (3.64 MB/s) - 'opencl_runtime_16.1.2_x64_rh_6.4.0.37.tgz' saved [117485559/117485559]

packer build base.json
virtualbox-iso output will be in this color.

==> virtualbox-iso: Downloading or copying ISO
    virtualbox-iso: Downloading or copying: http://cdimage.ubuntu.com/ubuntu/releases/bionic/release/ubuntu-18.04-server-amd64.iso
    virtualbox-iso: Download progress: 2%
    virtualbox-iso: Download progress: 5%
    virtualbox-iso: Download progress: 8%
    virtualbox-iso: Download progress: 11%
    virtualbox-iso: Download progress: 14%
    virtualbox-iso: Download progress: 17%
    virtualbox-iso: Download progress: 20%
    virtualbox-iso: Download progress: 21%
    virtualbox-iso: Download progress: 23%
    virtualbox-iso: Download progress: 26%
    virtualbox-iso: Download progress: 28%
    virtualbox-iso: Download progress: 31%
    virtualbox-iso: Download progress: 34%
    virtualbox-iso: Download progress: 37%
    virtualbox-iso: Download progress: 40%
    virtualbox-iso: Download progress: 44%
    virtualbox-iso: Download progress: 47%
    virtualbox-iso: Download progress: 51%
    virtualbox-iso: Download progress: 54%
    virtualbox-iso: Download progress: 58%
    virtualbox-iso: Download progress: 61%
    virtualbox-iso: Download progress: 65%
    virtualbox-iso: Download progress: 69%
    virtualbox-iso: Download progress: 72%
    virtualbox-iso: Download progress: 75%
    virtualbox-iso: Download progress: 79%
    virtualbox-iso: Download progress: 83%
    virtualbox-iso: Download progress: 86%
    virtualbox-iso: Download progress: 90%
    virtualbox-iso: Download progress: 93%
    virtualbox-iso: Download progress: 95%
    virtualbox-iso: Download progress: 97%
    virtualbox-iso: Download progress: 99%
==> virtualbox-iso: Starting HTTP server on port 8223
==> virtualbox-iso: Creating virtual machine...
==> virtualbox-iso: Creating hard drive...
==> virtualbox-iso: Creating forwarded port mapping for communicator (SSH, WinRM, etc) (host port 3144)
==> virtualbox-iso: Executing custom VBoxManage commands...
    virtualbox-iso: Executing: modifyvm instant-gnuradio-base --memory 4096
    virtualbox-iso: Executing: modifyvm instant-gnuradio-base --vram 128
    virtualbox-iso: Executing: modifyvm instant-gnuradio-base --cpus 2
    virtualbox-iso: Executing: modifyvm instant-gnuradio-base --natpf1 guestssh,tcp,127.0.0.1,2222,,22
    virtualbox-iso: Executing: modifyvm instant-gnuradio-base --accelerate3d on
    virtualbox-iso: Executing: modifyvm instant-gnuradio-base --usb on
    virtualbox-iso: Executing: modifyvm instant-gnuradio-base --usbxhci on
    virtualbox-iso: Executing: modifyvm instant-gnuradio-base --clipboard bidirectional
    virtualbox-iso: Executing: modifyvm instant-gnuradio-base --audio pulse
==> virtualbox-iso: Error executing command: VBoxManage error: VBoxManage: error: Invalid --audio argument 'pulse'
==> virtualbox-iso: Deregistering and deleting VM...
==> virtualbox-iso: Deleting output directory...
Build 'virtualbox-iso' errored: Error executing command: VBoxManage error: VBoxManage: error: Invalid --audio argument 'pulse'

==> Some builds didn't complete successfully and had errors:
--> virtualbox-iso: Error executing command: VBoxManage error: VBoxManage: error: Invalid --audio argument 'pulse'

==> Builds finished but no artifacts were created.
make: *** [vms/base/instant-gnuradio-base.ova] Error 1
`

bastibl, 2018-07-05:
Can you delete [this line](https://github.com/bastibl/instant-gnuradio/blob/master/base.json#L55) and try again, please.

bastibl, 2018-08-10:
The audio driver is no longer set explicitly, but the default for the architecture is used. Hopefully, that defaults to pulse on Linux and Coreaudio on OSX. https://github.com/bastibl/instant-gnuradio/commit/6ca06bad971ebfa3ea076231a41ff4c587e1fc74

closed 2018-08-10 by bastibl

deusama, 2018-08-22:
Sorry for my late response, I must have missed some mail from github.

Yes, deleting the audioout line helps. Thanks!