bastibl/instant-gnuradio #12: VM propagates your git config wherever it's built ;)

jkbecker, 2018-07-24:
Hi Basti,

we are using your VM to have a consistent baseline for our research. Out of convenience, we started committing code to our own, private project *from within* the VM and noticed that due to the lines linked below, any `instant-gnuradio` instance will automatically assume your github account credentials. You were a top 3 contributor to our private project for a second there, at least according to github stats ;)

I don't know how you want the build process to handle this, but I would suggest one or a combination of the following:

* build process grabs the git config from the host machine and configures the `instant-gnuradio` instance the accordingly
* build process prompts the user for git setup
* build process accepts command line arguments for git setup, so that it doesn't have to be interactive at build time
* build process leaves git setup empty (I'd prefer all other options over this)

Here is the culprit:
https://github.com/bastibl/instant-gnuradio/blob/4fad0a3094fdbb2b6ec2e6673a5c6a683562dc83/assets/gitconfig#L2

bastibl, 2018-07-24:
Hmm, it's unavoidable to edit the config and put the name. What would you suggest?
I could remove the name, then you get an error. I could put jon Doe, but that's not different then putting Bastian Bloessl :-)

bastibl, 2018-07-24:
> build process leaves git setup empty (I'd prefer all other options over this)

In my opinion that would be the only alternative. I think it is a bad idea to use the config of the machine that builds the VM. I guess only 0.00001% of the users want to go through the hassle and build the VM. I expect the main use-case to be that someone downloads the OVA or gets a copy during a workshop or something like that.

What do you think? I'd be happy to delete name and email from the config, so that an error is raised and users are pointed to the missing config, instead of a default that's wrong for all but one person on earth xD

bastibl, 2018-08-10:
I pushed a patch to leave the git user unset https://github.com/bastibl/instant-gnuradio/commit/6545cc20e2e54ca257750c8bfe8b4dc7bddcb7ff.

closed 2018-08-10 by bastibl