bastibl/instant-gnuradio #14: GNURadio Build Error

cyrus104, 2018-08-24:
When the process gets to gnuradio, it's throwing an error.

bastibl, 2018-08-24:
That's a bug of GNU Radio. The maintainer is aware of the issue and is working on a fix. 

http://lists.gnu.org/archive/html/discuss-gnuradio/2018-08/msg00158.html

closed 2018-08-24 by bastibl

cyrus104, 2018-08-26:
This issue was partially a GMP / MPIR missing issue. I added the following to get it to compile.
scripts/setup-gnuradio line 45
### GMP
pybombs -v install gmp

bastibl, 2018-08-27:
You're right. It's indeed yet another PyBombs bug. Since recently GNU Radio depends on GMP and the PyBombs receipt is not updated.
I just pushed a commit that does as you suggest https://github.com/bastibl/instant-gnuradio/commit/83091f2e61c25c27c2a70d76618b729808cd61bd

I'm not sure how to handle things like that in general. The script mutates into a big workaround for bugs in PyBombs and OOTs. It would be great if you could open an issue (or make a pull request) at gr-receipts so that this could be fixed upstream.

https://github.com/gnuradio/gr-recipes/blob/master/gnuradio.lwr#L20-L40