bastibl/instant-gnuradio #16: instant-gnuradio build fails during bladeRF configuration

jkbecker, 2019-02-05:
Hi Basti,

I just rebuilt the VM from the most recent master branch, and I'm getting this (full install log linked, pointing to the error):
https://gist.github.com/jkbecker/9a9a8f6ad65a982ec2e09b71ec7e464d#file-make-log-L1227

I'll look into it in the meantime, but maybe it's a very obvious issue to you...?

Cheers, Johannes

bastibl, 2019-02-05:
I'm also rebuilding atm. Nuand split udev rules in several files
```
sed 's/@BLADERF_GROUP@/plugdev/g' pybombs/src/bladeRF/host/misc/udev/88-nuand-bladerf1.rules.in | sudo tee pybombs/src/bladeRF/host/misc/udev/88-nuand-bladerf1.rules
sed 's/@BLADERF_GROUP@/plugdev/g' pybombs/src/bladeRF/host/misc/udev/88-nuand-bladerf2.rules.in | sudo tee pybombs/src/bladeRF/host/misc/udev/88-nuand-bladerf2.rules
sed 's/@BLADERF_GROUP@/plugdev/g' pybombs/src/bladeRF/host/misc/udev/88-nuand-bootloader.rules.in | sudo tee pybombs/src/bladeRF/host/misc/udev/88-nuand-bootloader.rules
```

jkbecker, 2019-02-05:
Sweet. And an asterisk in the line after as well, right?

    sudo cp pybombs/src/bladeRF/host/misc/udev/88-nuand*.rules /etc/udev/rules.d/

bastibl, 2019-02-05:
This line needs to be deleted. It directly pipes it to /etc

jkbecker, 2019-02-05:
Wait, I'm confused... shouldn't these lines read as follows then (tee'ing to /etc)?

    ### BLADERF
    pybombs -v install bladeRF
    sed 's/@BLADERF_GROUP@/plugdev/g' pybombs/src/bladeRF/host/misc/udev/88-nuand-bladerf1.rules.in | sudo tee /etc/udev/rules.d/88-nuand-bladerf1.rules
    sed 's/@BLADERF_GROUP@/plugdev/g' pybombs/src/bladeRF/host/misc/udev/88-nuand-bladerf2.rules.in | sudo tee /etc/udev/rules.d/88-nuand-bladerf2.rules
    sed 's/@BLADERF_GROUP@/plugdev/g' pybombs/src/bladeRF/host/misc/udev/88-nuand-bootloader.rules.in | sudo tee /etc/udev/rules.d/88-nuand-bootloader.rules

bastibl, 2019-02-05:
arg, you're right... I don't have a BladeRF to test :-/

jkbecker, 2019-02-05:
Me neither! I just want it to build, mostly ;)

bastibl, 2019-02-05:
OK :-) The other stuff worked for me earlier today.
I'm trying to add support for the Pluto and then  push the changes.

jkbecker, 2019-02-05:
Ah, cool!

Building without issues for me now.

bastibl, 2019-02-06:
Great, I'll close this then. Thanks!

closed 2019-02-06 by bastibl