bastibl/instant-gnuradio #19: Network Interface

gadjodilo83, 2019-03-19:
Hello

I dont have a Network Interface when i start the virtual machine with VMware.

How can i start it?

thanks 
gadjodilo83

bastibl, 2019-03-19:
I don't use VMWare, but after you imported the VM, you should be able to open it's settings and add network interfaces as needed.

closed 2019-03-29 by bastibl

alberto-fc, 2019-03-29:
Hi,

The naming scheme for network interfaces in VMware seems to differ from VBox (Debian 9 has introduced a new naming scheme for networks, no more generic eth**X**)

To be able to make the network work over VMware:

1. Edit /etc/netplan/01-netcfg.yaml ,, change the line below 'ethernets:' to your interface address (ens33 in my case, but it can differ in your case, check with 'ip link' how your iface is called)
2. Reboot or run 'netplan apply' in console

You shouldn't have any problem with the network anymore.

bastibl, 2019-03-29:
I think I didn't do any network-specific config and assumed things would "just work" when adding one :-)  At least it did for me. You should also be able to use the network manager (from the task bar) to configure the interface.

alberto-fc, 2019-03-29:
> I think I didn't do any network-specific config and assumed things would "just work" when adding one :-) At least it did for me. You should also be able to use the network manager (from the task bar) to configure the interface.

At least on the first boot, I couldn't list my interface from the UI (and the link was down) so I couldn't edit either the network parameters from the desktop.

_To be honest, maybe I missed some network panel on the GUI config, but I looked at every place and not luck._

bastibl, 2019-03-29:
I see. On VirtualBox it seems to be fine. Maybe VMWare creates a different interface that is affected by this bug:
https://askubuntu.com/questions/71159/network-manager-says-device-not-managed