bastibl/instant-gnuradio #20: [WISHLIST] Debian & Reduce size image

alberto-fc, 2019-03-27:
_First: Thank you; Nowadays it's hard to find something that work out-the-box. Even gnu-radio deleted their own VM links from their own wiki. Keeping this alive and updated is hats-off job._

I would love to see a shrink version of this with -**only**- the scripts to re-build the whole image from scratch to the full glory and size. I'm a bit biased against Ubuntu (last-releases, One doesn't like) and I try to avoid it.

Given that is unfair to ask for a fully-debian image after saying how complicated it is to keep it updated, I would rather to ask for an estimation about the effort which would require to keep an OS-agnostic scripts for OVA creation ( I didn't really dig on the whole process, is there any dependence on Ubuntu packages?  ).

My true intentions are fueled by _wishful thinking_ and being able to generate something architectural-agnostic from scratch. _My laptop is too heavy and I've a few ARMs asking for a job._

Thank you for sharing and your time.

bastibl, 2019-03-29:
I'm happy you appreciate the project. I can understand that Ubuntu isn't for everyone. The rationale was that I wanted to have the VM easy to use and easy to extend. And many software projects are well-tested with Ubuntu and I found the GNOME 3 desktop environment might be easy to use for someone who isn't necessarily a daily Linux user.

Regarding your query, I'm not sure if I understand you correctly. The VM is built with Packer but doesn't use any fancy tools for provisioning (like Ansible) but plain shell scripts:
https://github.com/bastibl/instant-gnuradio/tree/master/scripts
I guess 99% of these scripts should be similar for a Debian-based installation, but I didn't test. Maybe package names and install locations deviate.
One goals was also to provide a ready-to-use pre-configured desktop environment, i.e., it also configures links/shortcuts for relevant SDR applications. This, for sure, would have to be adapted for a different environment. What would be your intended use-case? Just have a bunch of shell scripts that can be run on a fresh Debian installation? Or a full workflow to build the VM?

Since you mentioned embedded devices, the way to go for these would be to use OpenEmbedded to build a firmware image:
https://github.com/balister/meta-sdr
But this is really a different story. I'm not sure how this could be integrated in instant-gnuradio, which is more about building VMs.

If you only want to install GNU Radio and some SDR applications on Debian or some ARM boards, then PyBombs would be the tool of choice. It's also used here for installation of the applications. That means, this project is not an installer for applications but a workflow to build a VM for SDR around them.

Regarding size, do you mind about the overhead from the desktop environment (i.e., Ubuntu) or about the SDR stuff. I was not sure about the latter. Either I leave git repositories and binaries of the build process on the VM or delete them after the VM is built. It depends mainly on the use-case as well. If one wants to just use the applications, it would save quite some bytes. If one wants to develop/recompile/extend applications, it's very handy to have repositories and binaries for incremental builds. Recently, I decided to delete build artifacts, so the current build is a bit smaller:
https://github.com/bastibl/instant-gnuradio/blob/master/scripts/setup-gnuradio.sh#L137-L141

alberto-fc, 2019-03-29:
Hi,

Thank you for the fast response. Shame on me, I didn't checked for meta-layers for OE. I have a fresh environment for a recent project with an ARM, I'll check the recipes for sure.

In reference to your questions, I'll try to summarize everything in few words: The main use case was/is to separate the tool generation from the VM generation, that allows to generate the environment from scratch (cross-compiling sometimes is pain, slow and you can use fancy tools like dist-cc to launch the process on a faster CPU).

Clearly, I missed a few points on your build generation, I'm going to re-check again with more time on my side with this extra info (and of course, I'll give a try to the metalayer)

By the way, regarding to the size question, I speak out of time, I checked your last build and the size is pretty tight, you can cut a few bits removing some Ubuntu-fancy tools (Amazon app and so on), but as a whole the SDR tools and dependencies take more space than anything else (percentage-wise).

P.S: FYI It works just fine on VMware, the new naming system for network for debian-alike-systems avoid your network to work out-the-box with VMware (the interface, at least for me on VMware is ens33 -it's configured for DHCP on another iface-, you need to edit /etc/netplan/01-netcfg.yaml *sigh* and apply 'netplan apply') and still the tool-box works like a charm. Thank you.

closed 2019-03-29 by alberto-fc

bastibl, 2019-03-29:
Thanks for the feedback. I'll have a look at the VMWare bug next week.

When it's about a platform independent way to install SDR apps, then PyBombs might be tool of choice. My understanding is that it supports cross-compiling and native compiling on ARM. So if you don't want to build a complete distro/firmware, it might be the easier than OE.