bastibl/instant-gnuradio #21: Build issue with gr-foo

cyrus104, 2019-06-10:
virtualbox-ovf: PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
    virtualbox-ovf: PyBOMBS.monitor_process() - DEBUG - Return value: 1
    virtualbox-ovf: PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
    virtualbox-ovf: PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-foo:
    virtualbox-ovf: Configuration failed
    virtualbox-ovf: PyBOMBS.install_manager - ERROR - Error installing package gr-foo. Aborting.
    virtualbox-ovf: Install tree:
    virtualbox-ovf: |
    virtualbox-ovf: \- gr-foo
    virtualbox-ovf: Configuring: (100%) [=========================================================]
==> virtualbox-ovf: Unregistering and deleting imported VM...
==> virtualbox-ovf: Deleting output directory...
Build 'virtualbox-ovf' errored: Script exited with non-zero exit status: 1

==> Some builds didn't complete successfully and had errors:
--> virtualbox-ovf: Script exited with non-zero exit status: 1

==> Builds finished but no artifacts were created.
Makefile:13: recipe for target 'vms/gnuradio/instant-gnuradio.ova' failed
make: *** [vms/gnuradio/instant-gnuradio.ova] Error 1

bastibl, 2019-06-11:
I updated my modules to the new proposed development model (of GNU Radio), i.e., I use `maint-3.7` and `maint-3.8` branches for development. I made a pull request (https://github.com/gnuradio/gr-recipes/pull/150/) to update the build files, but it is not merged yet. So PyBombs still points to outdated/non-existent branches.

As a temporary solution you could let PyBombs point to my fork of the receipts: https://github.com/bastibl/gr-recipes

cyrus104, 2019-07-03:
I'm closing this one because your merge request went through.

closed 2019-07-03 by cyrus104