bastibl/instant-gnuradio #23: Request: adding srsLTE to the build

cyrus104, 2019-07-03:
sudo apt-get install cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev
apt-get install libsctp-dev lksctp-tools
git clone https://github.com/srsLTE/srsLTE.git
cd srsLTE
mkdir build
cd build
cmake ../
make
make test

Not sure if the Makefile supports and make install but that might be the last step.

bastibl, 2019-07-09:
it's hard to decide what to include and where to draw the line since it's too specific. I'd be OK to included it if there is someone who actually uses it.
If you think it makes sense, could you please test it and make a PR? There are also official packages for Ubuntu, which might be better suited.

bastibl, 2019-07-09:
Didn't notice that this was already part of the PR. It's merged. Thanks.

closed 2019-07-09 by bastibl

cyrus104, 2019-07-09:
The version in there currently didn't work right at the very end.

I'll start looking at it.