bastibl/instant-gnuradio #31: Install as a host OS.

r00tb3, 2019-09-17:
I have a Thinkpad T480 with 16GB ram and 1TB HDD that I'd like to completely dedicate to GNU-Radio, OpenBTS, srsLTE etc for all the RF and Telecom stuff, should I just run  [scripts](https://github.com/bastibl/instant-gnuradio/tree/master/scripts) on a fresh Ubuntu 18.04 or 16.04 system and it will work?

Regards.

bastibl, 2019-09-17:
I didn't try it, but I see no reason why it should not work. I'd just suggest to use Ubuntu 18.04, as the VM does.

r00tb3, 2019-09-18:
> I didn't try it, but I see no reason why it should not work. I'd just suggest to use Ubuntu 18.04, as the VM does.

Thanks for the response. I'll try and let you know.

r00tb3, 2019-09-18:
> I didn't try it, but I see no reason why it should not work. I'd just suggest to use Ubuntu 18.04, as the VM does.

Which scripts should I run as some scripts remove a lot of folders and stuff?.

I think init-base, setup-base, setup-gnuradio scripts would be enough to set up the host OS.

bastibl, 2019-09-18:
Yes, and you have to copy the assets, see `base.json` and `gnuradio.json` . Also the scripts assume that your user is called gnuradio.

catalinalb, 2019-09-22:
Hello I have a fresh install of ubuntu LTS 18.04. (username and password gnuradio) I have to install python and just run the scripts? Or something else first? How can I run bason.json and gnuradio.json? The only way I see it is with running the make command but for that I have to install the VMware. In the end this is what I need a fresh ubuntu with a full gnuradio. No virtual maschines or usb sticks.

Thank you

closed 2019-09-30 by r00tb3