bastibl/instant-gnuradio #32: error in process sentinel: Window #<window 3> too small for splitting

catalinalb, 2019-09-23:
Help Please!

    virtualbox-iso: Checking /home/gnuradio/.emacs.d/elpa/25.2/espresso-theme-20181025.826...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/espresso-theme-20181025.826/espresso-theme-autoloads.el...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/espresso-theme-20181025.826/espresso-theme-pkg.el...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/espresso-theme-20181025.826/espresso-theme.el...
    virtualbox-iso: Done (Total of 1 file compiled, 2 skipped)
    virtualbox-iso: (Spacemacs) --> installing package: eval-sexp-fu@spacemacs-editing... [62/264]
    virtualbox-iso: Setting ‘package-selected-packages’ temporarily since "emacs -q" would overwrite customizations
    virtualbox-iso: Contacting host: melpa.org:443
    virtualbox-iso: Generating autoloads for eval-sexp-fu.el...
    virtualbox-iso: Generating autoloads for eval-sexp-fu.el...done
    virtualbox-iso:
    virtualbox-iso: Checking /home/gnuradio/.emacs.d/elpa/25.2/eval-sexp-fu-20190109.809...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/eval-sexp-fu-20190109.809/eval-sexp-fu-autoloads.el...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/eval-sexp-fu-20190109.809/eval-sexp-fu-pkg.el...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/eval-sexp-fu-20190109.809/eval-sexp-fu.el...
    virtualbox-iso: Done (Total of 1 file compiled, 2 skipped)
    virtualbox-iso: (Spacemacs) --> installing package: evil@spacemacs-bootstrap... [63/264]
    virtualbox-iso: Setting ‘package-selected-packages’ temporarily since "emacs -q" would overwrite customizations
    virtualbox-iso: (Spacemacs) --> installing package: evil-anzu@spacemacs-evil... [64/264]
    virtualbox-iso: (Spacemacs) --> installing package: evil-args@spacemacs-evil... [65/264]
    virtualbox-iso: (Spacemacs) --> installing package: evil-ediff@spacemacs-evil... [66/264]
    virtualbox-iso: (Spacemacs) --> installing package: evil-escape@spacemacs-base... [67/264]
    virtualbox-iso: Setting ‘package-selected-packages’ temporarily since "emacs -q" would overwrite customizations
    virtualbox-iso: Generating autoloads for evil-escape.el...
    virtualbox-iso: Generating autoloads for evil-escape.el...done
    virtualbox-iso:
    virtualbox-iso: Checking /home/gnuradio/.emacs.d/elpa/25.2/evil-escape-20180910.1234...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/evil-escape-20180910.1234/evil-escape-autoloads.el...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/evil-escape-20180910.1234/evil-escape-pkg.el...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/evil-escape-20180910.1234/evil-escape.el...
    virtualbox-iso:
    virtualbox-iso: In toplevel form:
    virtualbox-iso: evil-escape.el:90:1:Error: Cannot open load file: No such file or directory, evil
    virtualbox-iso: Done (Total of 0 files compiled, 1 failed, 2 skipped)
    virtualbox-iso: (Spacemacs) --> installing package: evil-exchange@spacemacs-evil... [68/264]
    virtualbox-iso: (Spacemacs) --> installing package: evil-iedit-state@spacemacs-evil... [69/264]
    virtualbox-iso: (Spacemacs) --> installing package: evil-indent-plus@spacemacs-evil... [70/264]
    virtualbox-iso: (Spacemacs) --> installing package: evil-lisp-state@spacemacs-evil... [71/264]
    virtualbox-iso: (Spacemacs) --> installing package: evil-magit@git... [72/264]
    virtualbox-iso: Setting ‘package-selected-packages’ temporarily since "emacs -q" would overwrite customizations
    virtualbox-iso: Contacting host: melpa.org:443
    virtualbox-iso: Parsing tar file...
    virtualbox-iso: Parsing tar file...done
    virtualbox-iso: Extracting with-editor-20190715.2007/
    virtualbox-iso: Extracting with-editor-20190715.2007/with-editor.info
    virtualbox-iso: Extracting with-editor-20190715.2007/with-editor.el
    virtualbox-iso: Extracting with-editor-20190715.2007/with-editor-pkg.el
    virtualbox-iso: Extracting with-editor-20190715.2007/dir
    virtualbox-iso: Generating autoloads for with-editor-pkg.el...
    virtualbox-iso: Generating autoloads for with-editor-pkg.el...done
    virtualbox-iso: Generating autoloads for with-editor.el...
    virtualbox-iso: Generating autoloads for with-editor.el...done
    virtualbox-iso:
    virtualbox-iso: Checking /home/gnuradio/.emacs.d/elpa/25.2/with-editor-20190715.2007...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/with-editor-20190715.2007/with-editor-autoloads.el...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/with-editor-20190715.2007/with-editor-pkg.el...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/with-editor-20190715.2007/with-editor.el...
    virtualbox-iso: Done (Total of 1 file compiled, 2 skipped)
    virtualbox-iso: Setting ‘package-selected-packages’ temporarily since "emacs -q" would overwrite customizations
    virtualbox-iso: Contacting host: melpa.org:443
    virtualbox-iso: Generating autoloads for git-commit.el...
    virtualbox-iso: Generating autoloads for git-commit.el...done
    virtualbox-iso:
    virtualbox-iso: Checking /home/gnuradio/.emacs.d/elpa/25.2/git-commit-20190717.29...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/git-commit-20190717.29/git-commit-autoloads.el...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/git-commit-20190717.29/git-commit-pkg.el...
    virtualbox-iso: Compiling /home/gnuradio/.emacs.d/elpa/25.2/git-commit-20190717.29/git-commit.el...
    virtualbox-iso: Done (Total of 1 file compiled, 2 skipped)
    virtualbox-iso: Setting ‘package-selected-packages’ temporarily since "emacs -q" would overwrite customizations
    virtualbox-iso: Contacting host: melpa.org:443
    virtualbox-iso: Parsing tar file...
    virtualbox-iso: Parsing tar file...done
    virtualbox-iso: Extracting transient-20190905.1138/
    virtualbox-iso: Extracting transient-20190905.1138/transient.el
    virtualbox-iso: Extracting transient-20190905.1138/transient-pkg.el
    virtualbox-iso: Extracting transient-20190905.1138/transient.info
    virtualbox-iso: Extracting transient-20190905.1138/dir
    virtualbox-iso: Generating autoloads for transient-pkg.el...
    virtualbox-iso: Generating autoloads for transient-pkg.el...done
    virtualbox-iso: Generating autoloads for transient.el...
    virtualbox-iso: Generating autoloads for transient.el...done
    virtualbox-iso:
    virtualbox-iso: Loading async...
    virtualbox-iso: (Spacemacs) --> installing package: evil-matchit@spacemacs-evil... [73/264]
    virtualbox-iso: (Spacemacs) --> installing package: evil-mc@spacemacs-evil... [74/264]
    virtualbox-iso: (Spacemacs) --> installing package: evil-nerd-commenter@spacemacs-evil... [75/264]
    virtualbox-iso: Setting ‘package-selected-packages’ temporarily since "emacs -q" would overwrite customizations
    virtualbox-iso: error in process sentinel: Window #<window 3> too small for splitting
==> virtualbox-iso: Unregistering and deleting virtual machine...
==> virtualbox-iso: Deleting output directory...
Build 'virtualbox-iso' errored: Script exited with non-zero exit status: 255

==> Some builds didn't complete successfully and had errors:
--> virtualbox-iso: Script exited with non-zero exit status: 255

==> Builds finished but no artifacts were created.
Makefile:10: recipe for target 'vms/base/instant-gnuradio-base.ova' failed
make: *** [vms/base/instant-gnuradio-base.ova] Error 1
gnuradio@gnuradio:~/instant-gnuradio$

catalinalb, 2019-09-25:
Anyone? And idea? Something...

subu1962, 2019-09-27:
I had the same issue. This seems to be an issue with emacs. So I disabled/commented out the installation of spacemacs from scripts/setup-base.sh
It went further but now I am getting errors about running out of space. 

Subu

catalinalb, 2019-09-28:
Thank you so much Subu. It seems to work :)

==> virtualbox-iso: Gracefully halting virtual machine...
==> virtualbox-iso: Preparing to export machine...
    virtualbox-iso: Deleting forwarded port mapping for the communicator (SSH, WinRM, etc) (host port 3194)
==> virtualbox-iso: Exporting virtual machine...
    virtualbox-iso: Executing: export instant-gnuradio-base --output vms/base/instant-gnuradio-base.ova

subu1962, 2019-09-28:
I got confused out by these messages:
virtualbox-iso: ==> Zero out the free space to save space in the final image
virtualbox-iso: dd: error writing '/EMPTY': No space left on device
virtualbox-iso: 73282+0 records in
virtualbox-iso: 73281+0 records out
virtualbox-iso: dd exit code 1 is suppressed

But they are EXPECTED!

From scripts/cleanup.sh
echo '==> Zero out the free space to save space in the final image'
dd if=/dev/zero of=/EMPTY bs=1M  || echo "dd exit code $? is suppressed"
rm -f /EMPTY

Since no count is specified, dd continues to write to /EMPTY till there's no more space left :-)

BTW, the emacs error is discussed here:
https://debbugs.gnu.org/cgi/bugreport.cgi?bug=24193

HOWEVER, after the base appliance builds fine, I have issues with building the gnuradio ova
packer build --force gnuradio.json 
``==> Builds finished. The artifacts of successful builds are:
--> virtualbox-iso: VM files in directory: vms/base
packer build --force gnuradio.json 
virtualbox-ovf output will be in this color.

==> virtualbox-ovf: Retrieving OVF/OVA
==> virtualbox-ovf: Trying vms/base/instant-gnuradio-base.ova
==> virtualbox-ovf: Trying vms/base/instant-gnuradio-base.ova
==> virtualbox-ovf: vms/base/instant-gnuradio-base.ova => /subu/instant_gnuradio/instant-gnuradio/packer_cache/a3f7ce16c9e6033f5a2047451cb8f6fa42e3ef99.ova
==> virtualbox-ovf: Importing VM: /subu/instant_gnuradio/instant-gnuradio/packer_cache/a3f7ce16c9e6033f5a2047451cb8f6fa42e3ef99.ova
==> virtualbox-ovf: Error importing VM: VBoxManage error: 0%...10%...20%...30%...40%...50%...60%...70%...80%...90%...100%
==> virtualbox-ovf: Interpreting /subu/instant_gnuradio/instant-gnuradio/packer_cache/a3f7ce16c9e6033f5a2047451cb8f6fa42e3ef99.ova...
==> virtualbox-ovf: OK.
==> virtualbox-ovf: 0%...
==> virtualbox-ovf: Progress state: NS_ERROR_INVALID_ARG
==> virtualbox-ovf: VBoxManage: error: Appliance import failed
==> virtualbox-ovf: VBoxManage: error: Code NS_ERROR_INVALID_ARG (0x80070057) - Invalid argument value (extended info not available)
==> virtualbox-ovf: VBoxManage: error: Context: "RTEXITCODE handleImportAppliance(HandlerArg*)" at line 957 of file VBoxManageAppliance.cpp
==> virtualbox-ovf: Deleting output directory...
Build 'virtualbox-ovf' errored: Error importing VM: VBoxManage error: 0%...10%...20%...30%...40%...50%...60%...70%...80%...90%...100%
Interpreting /subu/instant_gnuradio/instant-gnuradio/packer_cache/a3f7ce16c9e6033f5a2047451cb8f6fa42e3ef99.ova...
OK.
0%...
Progress state: NS_ERROR_INVALID_ARG
VBoxManage: error: Appliance import failed
VBoxManage: error: Code NS_ERROR_INVALID_ARG (0x80070057) - Invalid argument value (extended info not available)
VBoxManage: error: Context: "RTEXITCODE handleImportAppliance(HandlerArg*)" at line 957 of file VBoxManageAppliance.cpp
==> Some builds didn't complete successfully and had errors:
--> virtualbox-ovf: Error importing VM: VBoxManage error: 0%...10%...20%...30%...40%...50%...60%...70%...80%...90%...100%
Interpreting /subu/instant_gnuradio/instant-gnuradio/packer_cache/a3f7ce16c9e6033f5a2047451cb8f6fa42e3ef99.ova...
OK.
0%...
Progress state: NS_ERROR_INVALID_ARG
VBoxManage: error: Appliance import failed
VBoxManage: error: Code NS_ERROR_INVALID_ARG (0x80070057) - Invalid argument value (extended info not available)
VBoxManage: error: Context: "RTEXITCODE handleImportAppliance(HandlerArg*)" at line 957 of file VBoxManageAppliance.cpp

==> Builds finished but no artifacts were created.
make: *** [Makefile:13: vms/gnuradio/instant-gnuradio.ova] Error 1

Sigh. Need to debug this now.

Subu

subu1962, 2019-09-28:
Interestingly outside of packer I can import this fine.

`➜  base git:(master) ✗ VBoxManage import -n instant-gnuradio-base.ova  
0%...10%...20%...30%...40%...50%...60%...70%...80%...90%...100%
Interpreting /subu/instant_gnuradio/instant-gnuradio/vms/base/instant-gnuradio-base.ova...
OK.
Disks:
  vmdisk1	83886080000	-1	http://www.vmware.com/interfaces/specifications/vmdk.html#streamOptimized	instant-gnuradio-base-disk001.vmdk	-1	-1	

Virtual system 0:
 0: Suggested OS type: "Ubuntu_64"
    (change with "--vsys 0 --ostype <type>"; use "list ostypes" to list all possible values)
 1: Suggested VM name "instant-gnuradio-base 1"
    (change with "--vsys 0 --vmname <name>")
 2: Suggested VM group "/"
    (change with "--vsys 0 --group <group>")
 3: Suggested VM settings file name "/home/subu/VirtualBox VMs/instant-gnuradio-base 1/instant-gnuradio-base 1.vbox"
    (change with "--vsys 0 --settingsfile <filename>")
 4: Suggested VM base folder "/home/subu/VirtualBox VMs"
    (change with "--vsys 0 --basefolder <path>")
 5: Number of CPUs: 4
    (change with "--vsys 0 --cpus <n>")
 6: Guest memory: 4096 MB
    (change with "--vsys 0 --memory <MB>")
 7: USB controller
    (disable with "--vsys 0 --unit 7 --ignore")
 8: Network adapter: orig NAT, config 3, extra slot=0;type=NAT
 9: IDE controller, type PIIX4
    (disable with "--vsys 0 --unit 9 --ignore")
10: IDE controller, type PIIX4
    (disable with "--vsys 0 --unit 10 --ignore")
11: Hard disk image: source image=instant-gnuradio-base-disk001.vmdk, target path=instant-gnuradio-base-disk001.vmdk, controller=9;channel=0
    (change target path with "--vsys 0 --unit 11 --disk path";
    disable with "--vsys 0 --unit 11 --ignore")
➜  base git:(master) ✗ 
`
Issue with packer?

subu1962, 2019-09-28:
Followed this post and got it built! I really appreciate the Open Source community and how people are so willing to share information.

https://www.paulotte.de/post/2019/packer-vs-virtualbox/

Subu

catalinalb, 2019-09-28:
Hello,

So I've got stuck at building gnuradio. I did follow the issue and added "### GMP
pybombs -v install gmp" to setup-gnuradio.sh between ### LIBAD9361 and ### IIO OSCILLOSCOPE (not sure if is line 45 there as it was suggested) still I have the following errors:

No package 'thrift' found

Python checking for numpy - not found

No package 'gmp' found

No package 'mpir' found

Could NOT find Sphinx

Did not find liborc and orcc, disabling orc support...

virtualbox-ovf: -- Checking for module 'orc-0.4 > 0.4.11'
    virtualbox-ovf: --   No package 'orc-0.4' found

I am still fighting with this build from 2-3 weeks now (on my spare time). I just want the full package of GNU radio so I can run it on a native machine not in VM. I would appreciate if you could share the ISO image or the OVA image so I can create the ISO... 

It's sad that Live GNU radio image has been retired. I am not a linux expert just a beginner but I find it difficult to install all the packages for GNU Radio since I tried and didn't succeed. gr-fosphor I only managed to run it once on a windows machine with the QT in trial version. 

Thank you all for the patience and support.
Catalin

subu1962, 2019-09-29:
I have similar issues. After the base ova is built successfully, when packer tries to build the next ova with gnuradio  I run into all sorts of problems with dependencies etc.

IF all you are trying to do is build ONE instance of gnuradio, it might be easier to just start with the base OVA and do all the steps yourself to get gnuradio installed. This project is perhaps not the right thing if all you need is one native instance of Ubuntu with gnuradio. I think the intent is to have a standalone VM that can be used in classrooms. 

See https://www.bastibl.net/instant-gnuradio/
 
That's what I am planning to do for now.

My plan:
Let's start with the Makefile. The build command invokes packer with gnuradio.json. Now look at gnuradio.json and look at the type:shell files. The first one you will see is:
```
{
      "type": "shell",
      "scripts": [
        "scripts/setup-gnuradio.sh"
      ]
    },
```
Now look at the commands in setup-gnuradio.sh and execute the commands yourself from the command line. That will give you more control over what you are trying to achieve rather fight with packer and its virtualbox-ovf builder.

Note that I am not the developer or maintainer of instant-gnuradio. I am a user just like you and regardless of the issues, appreciate the originator of this project for providing me insights into how this can be built. 

Yes, there will be hiccups and IMO, it will be great learning experience.

Once we understand how to build it MANUALLY, we can retrofit the packer build with the needed changes to be successful. Time consuming? Yes.

Hope that helps. I will share my findings as I make progress so they could be helpful for others.

Subu

bastibl, 2019-09-30:
I'm not sure what the problem is or if there is still a problem. The PyBombs installation seems to be really a problem of PyBombs and not really related to this particular VM. Is there still an issue with the VM?

subu1962, 2019-10-07:
The .ova file that you can download at https://www.fleark.de/instant-gnuradio.ova is fine. 

The issue is with building a new VM for updated GNURadio etc.  The issue is a lot of software mentioned in the setup-gnuradio.sh have not been updated to work with any versions greater than GNURadio 3.7 

For ex: 
LimeSDR 
See https://github.com/myriadrf/gr-limesdr/issues/37
https://github.com/myriadrf/gr-limesdr/issues/44

OSMOSDR:
https://osmocom.org/issues/3855

And then there's the confusion with Python 2 vs Python 3. 
The recipes almost all call for building from source and there is a lot of this source code not being touched to update compatibility with newer versions makes it a big issue.

If one really wants all this GNURadio related software they should just stick with the .ova and not bother building with packer.

Right now I am going through a step at a time and documenting which ones work with GNURadio version 3.8 or even the latest 3.9 from the GNURadio master branch and which ones don't.

subu1962, 2019-10-07:
Even gr-fosphor fails:
'''
    virtualbox-ovf: + pybombs install gr-fosphor
    virtualbox-ovf: PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
    virtualbox-ovf: PyBOMBS - INFO - PyBOMBS Version 2.3.3
    virtualbox-ovf: PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
    virtualbox-ovf: Reading package lists...
    virtualbox-ovf: Building dependency tree...
    virtualbox-ovf: Reading state information...
    virtualbox-ovf: The following additional packages will be installed:
    virtualbox-ovf: libglfw3 libvulkan-dev libvulkan1
    virtualbox-ovf: Suggested packages:
    virtualbox-ovf: libegl1-mesa | libegl1-x11
    virtualbox-ovf: The following NEW packages will be installed:
    virtualbox-ovf: libglfw3 libglfw3-dev libvulkan-dev libvulkan1
    virtualbox-ovf: 0 upgraded, 4 newly installed, 0 to remove and 4 not upgraded.
    virtualbox-ovf: Need to get 1,359 kB of archives.
    virtualbox-ovf: After this operation, 12.1 MB of additional disk space will be used.
    virtualbox-ovf: Get:1 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 libglfw3 amd64 3.2.1-1 [49.2 kB]
    virtualbox-ovf: Get:2 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 libglfw3-dev amd64 3.2.1-1 [30.9 kB]
    virtualbox-ovf: Get:3 http://us.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 libvulkan1 amd64 1.1.70+dfsg1-1ubuntu0.18.04.1 [93.4 kB]
    virtualbox-ovf: Get:4 http://us.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 libvulkan-dev amd64 1.1.70+dfsg1-1ubuntu0.18.04.1 [1,185 kB]
    virtualbox-ovf: debconf: unable to initialize frontend: Dialog
    virtualbox-ovf: debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
    virtualbox-ovf: debconf: falling back to frontend: Readline
    virtualbox-ovf: debconf: unable to initialize frontend: Readline
    virtualbox-ovf: debconf: (This frontend requires a controlling tty.)
    virtualbox-ovf: debconf: falling back to frontend: Teletype
    virtualbox-ovf: dpkg-preconfigure: unable to re-open stdin:
    virtualbox-ovf: Fetched 1,359 kB in 1s (1,760 kB/s)
    virtualbox-ovf: Selecting previously unselected package libglfw3:amd64.
    virtualbox-ovf: (Reading database ... 196419 files and directories currently installed.)
    virtualbox-ovf: Preparing to unpack .../libglfw3_3.2.1-1_amd64.deb ...
    virtualbox-ovf: Unpacking libglfw3:amd64 (3.2.1-1) ...
    virtualbox-ovf: Selecting previously unselected package libglfw3-dev:amd64.
    virtualbox-ovf: Preparing to unpack .../libglfw3-dev_3.2.1-1_amd64.deb ...
    virtualbox-ovf: Unpacking libglfw3-dev:amd64 (3.2.1-1) ...
    virtualbox-ovf: Selecting previously unselected package libvulkan1:amd64.
    virtualbox-ovf: Preparing to unpack .../libvulkan1_1.1.70+dfsg1-1ubuntu0.18.04.1_amd64.deb ...
    virtualbox-ovf: Unpacking libvulkan1:amd64 (1.1.70+dfsg1-1ubuntu0.18.04.1) ...
    virtualbox-ovf: Selecting previously unselected package libvulkan-dev:amd64.
    virtualbox-ovf: Preparing to unpack .../libvulkan-dev_1.1.70+dfsg1-1ubuntu0.18.04.1_amd64.deb ...
    virtualbox-ovf: Unpacking libvulkan-dev:amd64 (1.1.70+dfsg1-1ubuntu0.18.04.1) ...
    virtualbox-ovf: Setting up libvulkan1:amd64 (1.1.70+dfsg1-1ubuntu0.18.04.1) ...
    virtualbox-ovf: Setting up libvulkan-dev:amd64 (1.1.70+dfsg1-1ubuntu0.18.04.1) ...
    virtualbox-ovf: Setting up libglfw3:amd64 (3.2.1-1) ...
    virtualbox-ovf: Setting up libglfw3-dev:amd64 (3.2.1-1) ...
    virtualbox-ovf: Processing triggers for libc-bin (2.27-3ubuntu1) ...
    virtualbox-ovf: PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
    virtualbox-ovf: PyBOMBS.install_manager - INFO - Installing package: gr-fosphor
    virtualbox-ovf: Cloning into 'gr-fosphor'...
    virtualbox-ovf: PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
    virtualbox-ovf: -- Extracting version information from git describe...
    virtualbox-ovf: -- Boost version: 1.65.1
    virtualbox-ovf: -- Found the following Boost libraries:
    virtualbox-ovf: --   system
    virtualbox-ovf: --   thread
    virtualbox-ovf: --   chrono
    virtualbox-ovf: --   date_time
    virtualbox-ovf: --   atomic
    virtualbox-ovf: CMake Error at CMakeLists.txt:124 (find_package):
    virtualbox-ovf: -- Configuring incomplete, errors occurred!
    virtualbox-ovf: Could not find a configuration file for package "Gnuradio" that is
    virtualbox-ovf: See also "/home/gnuradio/pybombs/src/gr-fosphor/build/CMakeFiles/CMakeOutput.log".
    virtualbox-ovf: compatible with requested version "3.7.3".
    virtualbox-ovf: See also "/home/gnuradio/pybombs/src/gr-fosphor/build/CMakeFiles/CMakeError.log".
    virtualbox-ovf:
    virtualbox-ovf: The following configuration files were considered but not accepted:
    virtualbox-ovf:
    virtualbox-ovf: /home/gnuradio/pybombs/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.0.0
    virtualbox-ovf:
    virtualbox-ovf:
    virtualbox-ovf:
    virtualbox-ovf: PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
    virtualbox-ovf: PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-fosphor:
    virtualbox-ovf: Configuration failed
    virtualbox-ovf: PyBOMBS.install_manager - ERROR - Error installing package gr-fosphor. Aborting.
    virtualbox-ovf: Install tree:
    virtualbox-ovf: |
[... cut ...]
Build 'virtualbox-ovf' errored: Script exited with non-zero exit status: 1

==> Some builds didn't complete successfully and had errors:
--> virtualbox-ovf: Script exited with non-zero exit status: 1

==> Builds finished but no artifacts were created.
Makefile:13: recipe for target 'vms/gnuradio/instant-gnuradio.ova' failed
make: *** [vms/gnuradio/instant-gnuradio.ova] Error 1
'''
As you can see gr-fosphor is compatible with GNURadio 3.7.3 only and fails to build with a more recent GNURadio version.

bastibl, 2019-10-07:
Yes. Unfortunately, there are many modules and projects that are not yet ported. Maybe you could document your work in an issue on PyBombs.

bastibl, 2019-12-17:
PyBombs is absolutely broken atm. Since they switched to the new GR branch, but only few modules and recipes are updated. I'll close this for now. I think it's PyBombs that should be fixed. Atm the VM build is pretty much broken :-/

closed 2019-12-17 by bastibl

catalinalb, 2019-12-17:
It's true. Can someone point to a compete docker image with the gnuradio modules and especially gr-fosphor?

bastibl, 2019-12-17:
sorry, no idea. I guess these images used PyBombs as well. If you find something, let me know.