bastibl/instant-gnuradio #33: Error installing Pybombs

r00tb3, 2019-09-30:
I'm facing the following error when I run setup-gnuradio.sh script.

```
./setup-gnuradio-new.sh    
+ export LANG=en_US.UTF-8
+ LANG=en_US.UTF-8
+ export PATH=/home/gnuradio/.local/bin:/home/gnuradio/bin:/home/gnuradio/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+ PATH=/home/gnuradio/.local/bin:/home/gnuradio/bin:/home/gnuradio/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+ sudo apt-get update
Ign:1 http://dl.google.com/linux/chrome/deb stable InRelease
Hit:2 http://in.archive.ubuntu.com/ubuntu xenial InRelease                          
Hit:3 http://archive.canonical.com/ubuntu xenial InRelease                                                          
Hit:4 http://ppa.launchpad.net/git-core/ppa/ubuntu xenial InRelease                                                 
Hit:5 http://in.archive.ubuntu.com/ubuntu xenial-updates InRelease                            
Hit:6 http://in.archive.ubuntu.com/ubuntu xenial-security InRelease     
Hit:7 http://dl.google.com/linux/chrome/deb stable Release
Reading package lists... Done                     
+ sudo apt -y install ipython python-matplotlib python-ipython-genutils python-scipy python-numpy python-pip python-qwt5-qt4 python-wxgtk3.0 multimon sox
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-numpy is already the newest version (1:1.11.0-1ubuntu1).
ipython is already the newest version (2.4.1-1).
multimon is already the newest version (1.0-7.1).
python-ipython-genutils is already the newest version (0.1.0-1).
python-matplotlib is already the newest version (1.5.1-1ubuntu1).
python-qwt5-qt4 is already the newest version (5.2.1~cvs20091107+dfsg-7build1).
python-scipy is already the newest version (0.17.0-1).
python-wxgtk3.0 is already the newest version (3.0.2.0+dfsg-1build1).
python-pip is already the newest version (8.1.1-2ubuntu0.4).
sox is already the newest version (14.4.1-5+deb8u4ubuntu0.1).
The following packages were automatically installed and are no longer required:
  libdbus-1-3:i386 libffi6:i386 libglib2.0-0:i386 libqwt-headers libreadline6:i386 snap-confine snapd-login-service
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 53 not upgraded.
+ pip install pybombs
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Requirement already satisfied: pybombs in /usr/local/lib/python2.7/dist-packages (2.3.3)
Requirement already satisfied: future in /usr/local/lib/python2.7/dist-packages (from pybombs) (0.17.1)
Requirement already satisfied: ruamel.yaml in /usr/local/lib/python2.7/dist-packages (from pybombs) (0.16.5)
Requirement already satisfied: setuptools in /usr/lib/python2.7/dist-packages (from pybombs) (20.7.0)
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from pybombs) (1.10.0)
Requirement already satisfied: requests in /usr/lib/python2.7/dist-packages (from pybombs) (2.9.1)
Requirement already satisfied: ruamel.yaml.clib>=0.1.2; platform_python_implementation == "CPython" and python_version < "3.8" in /usr/local/lib/python2.7/dist-packages (from ruamel.yaml->pybombs) (0.1.2)
Requirement already satisfied: ruamel.ordereddict; platform_python_implementation == "CPython" and python_version <= "2.7" in /usr/local/lib/python2.7/dist-packages (from ruamel.yaml->pybombs) (0.4.14)
+ pybombs -v recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /home/gnuradio/.pybombs/config.yml
Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 6, in <module>
    from pybombs.main import main
  File "/usr/local/lib/python2.7/dist-packages/pybombs/main.py", line 26, in <module>
    from pybombs.commands import dispatch
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/__init__.py", line 23, in <module>
    from .base import CommandBase, SubCommandBase, dispatch
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/base.py", line 27, in <module>
    from pybombs.config_manager import config_manager
  File "/usr/local/lib/python2.7/dist-packages/pybombs/config_manager.py", line 707, in <module>
    config_manager = ConfigManager()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/config_manager.py", line 320, in __init__
    self.load(select_prefix)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/config_manager.py", line 361, in load
    if self._append_cfg_from_file(self.local_cfg):
  File "/usr/local/lib/python2.7/dist-packages/pybombs/config_manager.py", line 445, in _append_cfg_from_file
    cfg_data = PBConfigFile(cfg_filename).get()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/config_file.py", line 87, in __init__
    raise PBException("Error loading {0}: {1}".format(filename, str(e)))
pybombs.pb_exception.PBException: Error loading /home/gnuradio/.pybombs/config.yml: while constructing a mapping
  in "<byte string>", line 11, column 7:
          gitrev: master
          ^ (line: 11)
found duplicate key "vars" with value "ordereddict([('config_opt', ' -DWITH_IIOD:BOOL=OFF -DINSTALL_UDEV_RULE:BOOL=OFF ')])" (original value: "ordereddict([('config_opt', ' -DWITH_IIOD:BOOL=OFF -DINSTALL_UDEV_RULE:BOOL=OFF ')])")
  in "<byte string>", line 16, column 7:
          vars:
          ^ (line: 16)

To suppress this check see:
    http://yaml.readthedocs.io/en/latest/api.html#duplicate-keys

Duplicate keys will become an error in future releases, and are errors
by default when using the new API.

```

I'm not able to fix the duplicate keys.

Pls, help.

Regards.

bastibl, 2019-09-30:
I don't know what you are doing, but it doesn't seem like you are just running the normal build of the VM. Sorry, but I cannot help you with this. I'd assume it is either an error converting the VM build to something else or a problem with PyBombs.

closed 2019-09-30 by bastibl

r00tb3, 2019-09-30:
I figured out the problem.

In the following commands i had to comment the echo -e line:
```
#### LIBIIO
pybombs config --package libiio gitrev master
#echo -e "      vars:\n        config_opt: ' -DWITH_IIOD:BOOL=OFF -DINSTALL_UDEV_RULE:BOOL=OFF '" >> /home/gnuradio/.pybombs/config.yml
pybombs -v install libiio
sudo mv 53-adi-plutosdr-usb.rules /etc/udev/rules.d/
```

In the current builld of libiio they've added the line into the config.yml

bastibl, 2019-09-30:
You're right.... I also just tried to rebuild and now gr-lime also fails. It's really impossible to maintain this project :-/ GNU Radio should really integrate PyBombs and core modules into its CI infrastructure.