bastibl/instant-gnuradio #34: Error in importing .OVA file in VMware workstation 12 pro

Ghaz75, 2019-12-15:
My vmware is 64bit and so is my host OS windows 10 home 64 bit. I downloaded the .ova file. I was unable to open it in vmware coz it showed some sha1 errors. I checked its crc it was fine, Then I extracted it using winrar. I got .mf , .ovf and .vmdk files. I tried to import using .ovf it booted up with grub rescue prompt. I set prefix to (hd0,msdos1)/boot/grub but insmod command doesn not find any linux or normal. My thiat attempt failed.
Then I tried creating an ubuntu virtual machine firstly 32 bit and then 64 bit and directing the harddisk to vmdk file. It boots up but shows readonly error for a partition in the vmdk file and when I either deny all or allow all to override that partition access it boots up with lot of errors and then stuck up at showing multiple warnings of Read only file system.   Seriously it is over-engineered :). Please help.

bastibl, 2019-12-17:
Can you please try to use the VM with VirtualBox. It's build with VirtualBox and it would be great to know if it's a problem of the VM or of the VMWare import.

Ghaz75, 2019-12-18:
Hello Dear. Thanks for your reply. I downloaded Oracle Virtualbox and tried to import your .ova file as appliance. Vbox gives this error ....  
 
Result Code: 
E_INVALIDARG (0x80070057)

Ghaz75, 2019-12-18:
Although while importing vbox shows the values of the parameters correctly e.g Ubuntu version, number of processors 2, RAM 4GB etc but once I click import, it gives failed to import and the above mentioned error code in detail. There is no error in the .ova file as far as CRC is concerned. Winrar extracts its vmdk, ovf and mf files correctly. Any idea please?

bastibl, 2019-12-18:
I downloaded the OVA again and imported it in VirtualBox 6.0.6 (with the extension pack installed). It works without problems. Can you please make sure that you have the extension pack installed and use a more or less recent version. Also is there no further information than this error?

Ghaz75, 2019-12-18:
Thanks Dear. I installed extension pack. And then re-run vbox and imported the .ova file. It gave the same error. Then I googled for the same error and I found that the operating system has to be restarted after the installation of extn pack and I did it. Now the VM is working fine. Thank you very much. Problem solved.

bastibl, 2019-12-19:
Great, I'll close the issue.

closed 2019-12-19 by bastibl