bastibl/instant-gnuradio #36: Feature Request - DSPECTRUM

cyrus104, 2019-12-29:
I wasn't sure if you have seen this but in testing it''s been very useful and I wanted to point it out.
I would like to ask that it to be included. 

There is a wiki to add it that wasn't very complicated.

[https://github.com/tresacton/dspectrumgui](https://github.com/tresacton/dspectrumgui)

bastibl, 2020-01-02:
Great to see that there is still interest in the VM. Unfortunately, it's, at the moment, not maintainable. I dumped lots of hours into this, but PyBombs is very unstable at the moment, i.e., the build fails every other day  for another reason. We really need to integrate PyBombs into the continuous integration framework of GNU Radio.
If you manage to build it (and add dspectrum) feel free to open a PR.
I'll leave the issue open, I hope that PyBombs stabilizes again and we can extend and update the VM again.

cyrus104, 2020-01-02:
Thanks for the message, seems like pybombs had issues since it came out. I
wish there was more focus on it.

bastibl, 2022-03-28:
I'll close that. If someone works on it, I'll be happy to review and merge.

closed 2022-03-28 by bastibl