bastibl/instant-gnuradio #37: Build issue gr-gsm osmocom

thusoo, 2020-01-10:
~/gr-gsm/build$ cmake ..
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:45 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Git: /usr/bin/git (found version "2.17.1") 
-- Extracting version information from git describe...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig3.0  
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'volk'
--   Found volk, version 2.0
-- Found VOLK: /usr/lib/x86_64-linux-gnu/libvolk.so  
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- Found CPPUNIT: /usr/lib/x86_64-linux-gnu/libcppunit.so;dl  
-- Checking for module 'libosmocore'
--   No package 'libosmocore' found
-- Could NOT find libosmocore (missing: LIBOSMOCORE_LIBRARY LIBOSMOCORE_INCLUDE_DIR) 
-- Checking for module 'libosmocodec'
--   No package 'libosmocodec' found
-- Could NOT find libosmocodec (missing: LIBOSMOCODEC_LIBRARY LIBOSMOCODEC_INCLUDE_DIR) 
-- Checking for module 'libosmocoding'
--   No package 'libosmocoding' found
-- Could NOT find libosmocoding (missing: LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR) 
-- Checking for module 'libosmogsm'
--   No package 'libosmogsm' found
-- Could NOT find libosmogsm (missing: LIBOSMOGSM_LIBRARY LIBOSMOGSM_INCLUDE_DIR) 
CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/home/shongpan/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/home/shongpan/gr-gsm/build/CMakeFiles/CMakeError.log".
i tried troubleshooting each error or missing lib/object manually but still doesn't make.

bastibl, 2020-01-10:
That's not a problem of this VM but stems from the fact that PyBombs is totally broken since the 3.8 release. It's atm not possible to maintain this project. See also the other issues, for example: https://github.com/bastibl/instant-gnuradio/issues/36

thusoo, 2020-01-15:
Thanks for the quick response @bastibl 
I can understand & hope with the PyBombs update the VM is smooth & up running :+1: 
Cheers!

closed 2020-01-15 by thusoo