bastibl/instant-gnuradio #39: No gr-video-sdl module in the tree list of grc

Ghaz75, 2020-04-06:
Dear Bastibl
Thanks for you wonderful image of gnuradio. Dear I have noticed that the subject block is missing from the tree list of grc. I understand that the gnuradio installed using pybombs has this issue which is not there if one builds it from the source. I am in need of this block for my research. Although the folder exists in pybombs/src/gnurado/ and even the cmakelist.txt file is also there in gr-video-sdl/grc , I also tried to compile it and it was compiled without throwing errors then I use make and sudo make install and it was installed yet once I open grc the video-sdl block is still missing. I searched from the forums and it is written that this block is added during compilation of complete build of gnuradio. Is there any other way to add it as OOT module/ block coz I dont want to break the entire build of gnuradio in your image. I shall be grateful if you give me any advice on it or compile it in next version of your virtual image. 
Thanks & Kind Regards
Ghazanfar

bastibl, 2020-04-07:
Hi, it looks like the video component is not enabled with PyBombs. But it just misses the `libsdl` dependency, which can be installed with `sudo apt install libsdl1.2-dev`. You can either add this to the build script when rebuilding the VM. Or start the existing VM, install SDL, and rebuild GNU Radio with
```
pybombs -p master rebuild gnuradio
```
I didn't wait until the build completed, but it should work :-) Let me know if there are problems.

Ghaz75, 2020-04-11:
Dear I use rebuild command in loaded vm's console. I could see the gr-video-sdl in the list to be compiled but after showing some gnuradio disabled components, it throws error as under:-
 Using install prefix: /home/gnuradio/pybombs
-- Building for version: 3.7.13.4 / 3.7.13.4
-- Configuring incomplete, errors occurred!
See also "/home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
PyBOMBS.Packager.source - ERROR - Configuration failed
PyBOMBS.rebuild - ERROR - Error rebuilding package gnuradio. Aborting.

I checked the cmake error log which is :-

Determining if the pthread_create exist failed with the following output:
Change Dir: /home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_68195/fast"
/usr/bin/make -f CMakeFiles/cmTC_68195.dir/build.make CMakeFiles/cmTC_68195.dir/build
make[1]: Entering directory '/home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_68195.dir/CheckSymbolExists.c.o
/usr/bin/cc   -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized    -o CMakeFiles/cmTC_68195.dir/CheckSymbolExists.c.o   -c /home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_68195
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_68195.dir/link.txt --verbose=1
/usr/bin/cc  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized     -rdynamic CMakeFiles/cmTC_68195.dir/CheckSymbolExists.c.o  -o cmTC_68195 
CMakeFiles/cmTC_68195.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_68195.dir/build.make:97: recipe for target 'cmTC_68195' failed
make[1]: *** [cmTC_68195] Error 1
make[1]: Leaving directory '/home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_68195/fast' failed
make: *** [cmTC_68195/fast] Error 2

File /home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_0e6e2/fast"
/usr/bin/make -f CMakeFiles/cmTC_0e6e2.dir/build.make CMakeFiles/cmTC_0e6e2.dir/build
make[1]: Entering directory '/home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_0e6e2.dir/CheckFunctionExists.c.o
/usr/bin/cc   -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_0e6e2.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.10/Modules/CheckFunctionExists.c
Linking C executable cmTC_0e6e2
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_0e6e2.dir/link.txt --verbose=1
/usr/bin/cc  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_0e6e2.dir/CheckFunctionExists.c.o  -o cmTC_0e6e2 -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_0e6e2.dir/build.make:97: recipe for target 'cmTC_0e6e2' failed
make[1]: *** [cmTC_0e6e2] Error 1
make[1]: Leaving directory '/home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_0e6e2/fast' failed
make: *** [cmTC_0e6e2/fast] Error 2

Determining if the include file intrin.h exists failed with the following output:
Change Dir: /home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp

[... 338 lines cut ...]
   return ((int*)(&Q_WS_MAC))[argc];
                   ^~~~~~~~
                   Q_WS_X11
CMakeFiles/cmTC_084bf.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_084bf.dir/CheckSymbolExists.cxx.o' failed
make[1]: *** [CMakeFiles/cmTC_084bf.dir/CheckSymbolExists.cxx.o] Error 1
make[1]: Leaving directory '/home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_084bf/fast' failed
make: *** [cmTC_084bf/fast] Error 2

File /home/gnuradio/pybombs/src/gnuradio/build/CMakeFiles/CMakeTmp/CheckSymbolExists.cxx:
/* */
#include <QtCore/qglobal.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef Q_WS_MAC
  return ((int*)(&Q_WS_MAC))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Sorry for long text.

Ghaz75, 2020-04-11:
And by using your given command I am afraid that the build folder was removed and new folder was created by pybombs which has this incomplete files. I have lost your original build. Any suggestion if you can give me or is there any way that if I can use an alternate module in place of video-sdl-link by replacing video-sdl-link box in grc flowgraph with that alternate module.

bastibl, 2020-04-13:
Sorry, but I have no idea why it doesn't work for you. I guess you modified the VM or broke it by trying out something.
I downloaded the VM from the link in the Readme, imported it in VirtualBox, and did as suggested in my previous post. It works and the Video SDR blocks show up in GRC.
Also, the VM doesn't contain build directories (they are deleted during the build process). If you have an older version, you can try to delete them before rebuilding.

Ghaz75, 2020-04-18:
Hello Dear. I made a new instance of the vm and rebuilt the gnuradio using the command pybombs -p master rebuild gnuradio. It was succesfully built up and now I have video-sdl-sink in the list. But gr-gsm is gone so please guide me should I build it using this command "pybombs -p master install gr-gsm" or should I install it without  "-p master" to run smoothly? Thanks

bastibl, 2020-04-20:
There shouldn't be any difference. There is only a `master` prefix. You don't have to specify it explicitly. I'm a bit worried that there will be issues, since PyBombs points to a branch and not to individual commits. And, for example, the `master` branch of the project might meanwhile point to a branch to is only compatible to GNU Radio 3.8. But you can give it a try. If it doesn't work, you have to compile from source.

bastibl, 2020-10-09:
I'll close this, since we switched to using GNU Radio from a PPA, i.e., do no longer compile from source with PyBombs.

closed 2020-10-09 by bastibl