bastibl/instant-gnuradio #40: Major update

cyrus104, 2020-04-07:
@bastibl 

I wanted to see if you are looking at updating this to GR3.8? I have been looking at jonpeck / instant-gnuradio which was updated earlier this year but it's not building right now.

I understand how much work you went through to make this. I have taken that one and have worked on some building it with 3.8 and getting a fair amount of SDRs working it... it's not building 100% yet.

If I get this to build successful once, I'll put in a request and maybe you can clean it up and fix any dumb things I've done.

cyrus104, 2020-04-07:
I'm currently building it for 19.10 but will look at changing to 20.04 tomorrow that should give us some stability.

It's very valuable to have a good way to rebuild a VM working with most of the tools already pre-installed.

bastibl, 2020-04-07:
Would be awesome, if you would update it to 3.8. I'm happy to review and merge your changes.

Ghaz75, 2020-04-11:
Nice suggestion, I also had been trying to build GR3.8 but in Kali Linux 2019. I am partially successful. I tried initially by installing it using pybombs but that version is a mess. Then I built it from its binary source. Although it works but the most important modules like gr-gsm is still not compatible with it. I tried different forks' branches on which people are working but still it needs a lot of changes due to xml to yml conversion in the main build of gr3.8. So far the best and working build as a full fledged vm image is this release for gr3.7 by Bastibl. I am grateful to you Bastibl.

Ghaz75, 2020-04-11:
> I'm currently building it for 19.10 but will look at changing to 20.04 tomorrow that should give us some stability.
> 
> It's very valuable to have a good way to rebuild a VM working with most of the tools already pre-installed.

And of course at the same time if the newest build is compatible with old modules then it would be a wonderful effort especially the compatibility of newest build with oldest modules like gr-baz.

cyrus104, 2020-04-11:
Man, you're right about pybombs. I really the idea of it but it's a lot of work and seems to keep changing. There is better documentation for using it in the github issues section than in the readme section. I have a post over there with a few errors that I was running into. It doesn't seem to heavily trafficed so I'm not sure it will get looked at.

I'm still working it going the source method but pybombs is nice for install stuff that isn't the core. It's weird though, one package is missed and even though there is a gr-recipe for it... it doesn't even attempt it.

cyrus104, 2020-04-12:
@bastibl @Ghaz75 
I saw a new OS build focused on SDRs called DragonOS. There is  Lubuntu 18.04LTS build and a Debian 10 build. Right now only the ISO is available but I have posted a question to the creator to highlight this project. I think it would be great if more people were able to focus around a single project, this project is nice because it's fully open and easy enough to modify. If the base-setup is done on a bare install, I'm not sure if we can have packer do the install on the local machine. The current Dragon is an ISO that can be used as an installer.

https://sourceforge.net/projects/dragonos-10/

bastibl, 2020-10-09:
Recently, I did a major update to provide a VM for GRCon20. The current version is based on Ubuntu 20.04 and uses a GNU Radio PPA to install a recent GNU Radio version.

closed 2020-10-09 by bastibl