bastibl/instant-gnuradio #42: Feature Request: Support Adalm Pluto in grc

eneukirchner, 2021-11-20:
Could you possible add support for Adalm Pluto SDR as a block in gnuradio-companion?

bastibl, 2021-11-22:
Unfortunately that's not so easy given the desperate packaging state. Ubuntu 20.04 comes with GNU Radio 3.8, but the gr-iio package is broken. At the moment, the VM uses a PPA. When it was built initially, this PPA had GNU Radio 3.8 with some compatible OOTs. If you rebuilt now, the PPA ships GNU Radio 3.9 but none of the OOTs are updated... I'm happy to update and rebuilt, but I want to avoid compiling OOTs from source (again) or using PyBombs. Maybe you could ask if someone would be motivated to update in-tree or PPA packages.

eneukirchner, 2021-11-23:
There is hope https://www.youtube.com/watch?v=2gKbollW6wg

bastibl, 2021-11-23:
Yes, but nothing that will happen soon :-/  GNU Radio 3.10 needs to be released and packaged, and then OOTs have to be ported to 3.10 and packaged. AFAIK, there will be some infrastructure work on GNU Radio, which might help to stream-line this process, but I'd assume that's >5 months. So, imho, it would still make sense to update the PPA.

danielgallagher, 2021-12-22:
My fork has a branch that adds PlutoSDR support linked below. It was buildable and functional earlier this year. Feel free to give it a try. I would be interested hearing if it still builds successfully. 

https://github.com/danielgallagher/instant-gnuradio/tree/ua_ece531

I am happy to create a PR if @bastibl wants to verify and merge the updates. I agree that this a temporary solution until GNU Radio 3.10 is ready.

bastibl, 2022-03-28:
Sorry, I missed this :-/ Now that 3.10 is out, it's probably better to go down that road.

closed 2022-03-28 by bastibl