bastibl/instant-gnuradio #43: Now that GNU Radio 3.10 is out...

drws, 2022-03-16:
GNU Radio 3.10 was [released two months ago](https://github.com/gnuradio/gnuradio/releases/tag/v3.10.0.0) and even though it cleaned up some things, [PyBOMBS is now unsupported](https://wiki.gnuradio.org/index.php?title=InstallingGR#Using_PyBOMBS). What does that mean for `instant-gnuradio`?

bastibl, 2022-03-16:
This project doesn't use PyBombs. The current version uses a PPA, but that broke because the GNU Radio PPA is not/was not maintained.

drws, 2022-03-16:
It would appear I was reading some [old info](https://github.com/bastibl/instant-gnuradio/issues/36#issuecomment-570221441) before. Thank you for clearing it up.

AFAIK the GNU Radio PPA never really was maintained for a noticeable amount of time.

Are there any plans to revive this project by employing the only `3.10`-supported build procedure (the manual one)?

bastibl, 2022-03-16:
Pull Requests are very welcome and I'm happy to help, but I don't have the bandwidth to do it myself at the moment.

bastibl, 2022-04-13:
The recent updates provide GNU Radio 3.10 through Ubuntu PPAs.

closed 2022-04-13 by bastibl