bastibl/instant-gnuradio #51: Possibly incorrect optimizations in 90-usrp.conf

drws, 2022-10-05:
The [`90-usrp.conf`](https://github.com/bastibl/instant-gnuradio/blob/8ea565d556b59811511b2c9d00f7aa52e99fbd44/assets/90-usrp.conf) file for setting kernel parameters in `/etc/sysctl.d/` contains 3 lines, only 2 of which are unique. Furthermore, it doesn't set `rmem_default` and `wmem_default` as per the [USRP docs](https://kb.ettus.com/USRP_Host_Performance_Tuning_Tips_and_Tricks#Adjust_Network_Buffers). I believe the contents of the file should be:

    net.core.rmem_default=33554432
    net.core.wmem_default=33554432
    net.core.rmem_max=33554432
    net.core.wmem_max=33554432

The value 33554432 means 32MiB whereas the current 50MB doesn't seem practical for a memory chunk size. Is there a reason a higher value than 32MiB is used, possibly because of some other SDR hardware than USRP?

bastibl, 2022-10-06:
I guess, there is no right or wrong optimization. Just one that works better or worse. I don't know what is better and also Ettus is not clear on that: https://github.com/EttusResearch/uhd/search?q=rmem_max&type=code. IIRC, I didn't make up this value but got also got it from the recommendation of the driver. I'm happy to merge 32MB, if you prefer that. Can you open a PR?

drws, 2022-10-11:
I'm not actually using `instant-gnuradio`, I was just checking out its code and left the comment since it's good to have working GR builds around. I also don't know what value would be best, so you can just leave it for now. But there's still the duplicated line and no setting of `*_default` values (which might be redundant).

commit cc0d9c7d (2022-10-12): update sysctl for usrp

fixes #51
closed 2022-10-12 by bastibl