bastibl/instant-gnuradio #52: Ubuntu updated to 22.04.2

ryanmtaylor, 2023-02-28:
Ubuntu was updated to 22.04.2 and so the links to 22.04.1 no longer work – I also couldn't get SSH to work in 22.04.2 – I can get the command line ssh running for the base image for 20.04.5 but the password won't work.

bastibl, 2023-02-28:
I just pushed some changes. Can you try the most recent version please?

bastibl, 2023-07-24:
Thanks for the info. Could you please make a PR with the version that works for you? Then, I can test w/ VirtualBox.

bastibl, 2023-08-07:
I'll close this, since the original issue is resolved (i.e., it uses Ubuntu 22.04.2). If there are still networking issues w/ vmware, please open a PR. Since I'm not sure where you added these lines.

closed 2023-08-07 by bastibl