bastibl/instant-gnuradio #54: Last build with GR3.8 availability (for legacy/archive purposes)

drws, 2023-09-07:
It would be great to have the latest "legacy" version (pre Ubuntu 22) available somewhere for legacy compatibility purposes. Unfortunately I only have some older version from around March 2022, but not the latest one (9 May 2022) if I'm reading the commit log correctly. If author or somebody else has the right vanilla OVA file, I kindly ask you to share. Or perhaps I missed it somewhere?

bastibl, 2023-09-08:
Please see the branches, e.g., `maint-3.8`.

bastibl, 2023-09-08:
I guess, old commits from years ago won't build anymore. PPAs got deleted. Git checks out the current head (i.e., are not pinned to a commit). ISOs no longer available or moved to archive URLs, same for apt repos. etc. etc. So I guess that's not practical or would require a volunteer to maintain.

drws, 2023-09-10:
You seem to be talking about version maintenance in the _future_. But this issue is **specifically about past** build(s) with GR3.8.

Do you possibly have any such builds in your archive?

bastibl, 2023-09-15:
GNU Radio hosts an old build: https://wiki.gnuradio.org/index.php?title=UbuntuVM